fritz-smh/yi-hack #231: 12CN always yellow on

n00d1e5, 2020-03-23:
Firmware: 1.8.7.0C_201705091058
Model: 12CN

---

**First case:**
- Format the whole TF-card
- Put test + home inside
-> yellow on without blink

**Second case:**
- Only format the main zone (formatted by the MI Home app)
- Put test + home inside
-> yellow on without blink

**Third case:**
- Without format
- Put test + home inside
-> yellow on with blink

*In first and second cases, there won't be record and record_sub folders.*

---

I find the issue [#212](https://github.com/fritz-smh/yi-hack/issues/212).
But the case is even I turn the debug on, there is no new file comes out (with 15min waiting while yellow on).

kruschman, 2020-03-23:
This is an abandoned project. I wouldn't expect it to work in any case.

n00d1e5, 2020-03-23:
Solved:

- Model: YHS-113 / 12CN

1. create folder "test" at root folder
2. create file "equip_test.sh" in "test"

```bash
#!/bin/sh
# Telnet
if [ ! -f "/etc/init.d/S88telnet" ]; then
    echo "#!/bin/sh" > /etc/init.d/S88telnet
    echo "telnetd &" >> /etc/init.d/S88telnet
    chmod 755 /etc/init.d/S88telnet
fi
dr=`dirname $0`
# fix bootcycle
mv $dr/equip_test.sh $dr/equip_test.sh.moved

echo "root:n00d1e5Yummy" | chpasswd
reboot
```

3. copy "sd\test\rtspsvrM" to root folder
4. insert the memory card back to camera
5. plug-in
6. from terminal, `telnet camera_ip`
7. `cp /tmp/hd1/rtspsvrM   /home/rtspsvr`

source: https://bbs.hassbian.com/thread-3402-1-1.html

closed 2020-03-25 by n00d1e5