fritz-smh/yi-hack #240: Wifi is propably connecting, but then orange led is blinking (check log)

kamilk91, 2022-05-30:
```
Starting to log...
Start telnet server...
MTK 7601
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.6.1Q_201607271501
    [kernel]
    version=3-0.8
    provider=1
    devtype=1
Firmware letter is : 'Q'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 May 30  2022 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 May 30  2022 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 May 30  2022 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 May 30  2022 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 May 30  2022 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:66:F4:CA  
              inet addr:192.168.1.63  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe66:f4ca/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:323 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:123454 (120.5 KiB)  TX bytes:200 (200.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
Previous datetime is Thu Jan  1 00:01:01 GMT 1970
Done
New datetime is Thu Jan  1 00:01:31 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1814 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1841 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1945 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  621 root         0 SWN  [jffs2_gcd_mtd4]
  625 root      1712 S    /bin/sh /etc/init.d/rcS
  637 root       872 S <  udevd --daemon
  645 root      1704 S    /bin/sh /etc/init.d/S80network
  659 root         0 SWN  [jffs2_gcd_mtd5]
  660 root      1708 S    /bin/sh ./init.sh
  667 root         0 SW   [flush-mtd-unmap]
  678 root      1716 S    /bin/sh /home/hd1/test/equip_test.sh
  680 root         0 SW   [flush-179:0]
  685 root      1704 S    telnetd
  689 root       984 S    /home/log_server
  824 root         0 SW   [RtmpTimerTask]
  825 root         0 SW   [RtmpMlmeTask]
  826 root         0 SW   [RtmpCmdQTask]
  827 root         0 SW   [RtmpWscTask]
  957 root         0 SW   [hidog]
  995 root      1164 S    ./peripheral
  996 root      1144 S <  ./dispatch
  997 root      1068 S    ./exnet
 1163 root      3132 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1814 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1841 root      1408 S    ./serverM 80
 1859 root      3360 S    ./record_event
 1860 root     14732 S    ./mp4record 60
 1889 root      1716 S    /usr/sbin/crond -b
 1938 root       904 S    /home/led_ctl -boff -yfast
 1939 root     94472 S    ./rmm
 1945 root     27252 S N  ./rtspsvrM
 2040 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.6M    904.0K  75% /
tmpfs                    17.2M     24.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.1M    800.0K  91% /home
tmpfs                    32.0M     48.0K     32.0M   0% /tmp
/dev/hd1                252.0M      8.4M    243.6M   3% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                252.0M      8.4M    243.6M   3% /tmp/hd1/test/http/record

```

it looks like its ok, but it isnt. after blue stoping blinking, there is about 5 seconds OK and its blue, but then orange starts blinking. logs shows not many.

albarzoida, 2023-12-12:
O

albarzoida, 2023-12-12:
> ```
> Starting to log...
> Start telnet server...
> MTK 7601
> Hardware version informations :
>     hwplatform=2
> Software version informations :
>     [sys]
>     version=1.8.6.1Q_201607271501
>     [kernel] [...]